richards-tech/RTIMULib-Arduino #11: MPU9150 and Arduino Leonardo

Dorian96, 2015-01-21:
Hi all!!
I've got this errorcode on Leonardo, can you help me?

ArduinoIMU starting using device MPU-9150
Failed to init IMU: -1
No valid compass calibration data

richardbarnett, 2015-01-22:
-1 means that the first write the MPU9150 failed. Usually this indicates an electrical problem. Check RTIMUMPU9150.cpp to see where these error codes are generated.

Dorian96, 2015-01-22:
Ok, i found that:

if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_PWR_MGMT_1, 0x80))
        return -1;

does It mean that it can't communicate with the sensor?

Dorian96, 2015-01-22:
I've connected it like:

Leonardo <-> Sensor
SCL  <->  SCL
SDA  <->  SDA
3.3V  <->  VDD
GND  <->  GND
2   <->   INT
But i receive the same error... WHY!?

richardbarnett, 2015-01-22:
The interrupt line should not be connected - it is not used and has caused problems in some cases. Since the Leonardo is an Uno with a USB interface effectively, it should be working if everything else is correct however the Leonardo is not one of the boards used for testing.

Well it looks like it should be working. One thing I can suggest is to use an oscilloscope or logic analyzer to see what's actually happening on the I2C lines. 

Another thing you might want to check is RAM utilization. The Leonardo does use more RAM for the USB interface and this may in fact be too much. The result can be a stack/heap crash with unpredictable results and could possibly explain what you are seeing.

kirilligum, 2015-03-01:
turned out in my case, i had to set the address of the sensor from 0x69 to 0x68. Also, I needed to solder it.

richardbarnett, 2015-03-01:
Glad it's now working.

closed 2015-03-01 by richardbarnett