richards-tech/RTIMULib-Arduino #12: MPU9150 with Arduino Nano Pro and Mega2560

iflavian, 2015-01-30:
Hello!

I have the following problem with MPU9150 - Arduino Nano Pro v3 and Mega2560:
............................................................................................................................
ArduinoIMU starting using device MPU-9150
Failed to init IMU: -1
No valid compass calibration data
...........................................................................................................................
I use the Drotek  IMU 9DOF - MPU9150 breakout board.

The I2C scanner reports the address: 0x69 for the IMU, so i don't think there is an electrical connection issue between IMU and Arduino.

I use the following electrical connection:
Arduino          IMU
        SCL <-> SCL
        SDA <-> SDA
3.3V (5V) <-> VDD  (-> I tried both 3.3V and 5V. Anyway the IMU board has a level convertor onboard)
        GND <-> GND

Can you help me with some advices? Or I have a defective IMU board? :(

richardbarnett, 2015-01-30:
It certainly implies an interface issue. Did you correctly select the 0x69 MPU-9150 in this file - https://github.com/richards-tech/RTIMULib-Arduino/blob/master/libraries/RTIMULib/RTIMULibDefs.h? The default is for 0x68 and would give the result you are seeing.

iflavian, 2015-01-30:
Thank you for your feedback.
Yes, I correctly selected the 0x69 address in RTIMULibDefs.h
I forgot to mention this in my previous post. Sorry...

richardbarnett, 2015-01-30:
Of course it would be sensible if I displayed the IMU address as well as the type when the program starts! Anyway, a return code of -1 means that the first I2C write to the IMU failed and this is almost always a hardware issue. If you have an oscilloscope or logic analyzer you could try looking at the I2C lines to make sure they are doing something reasonable.

iflavian, 2015-01-30:
Ok.
I'll use the logic analyzer to peek at the I2C signals.
I'll keep you informed.
Thanks

iflavian, 2015-02-01:
Hello,
Please find attached the result of I2C signals logic analyze.
I used logic analyzers from IKALogic and Saleae.
IKALogic:
[image]
Saleae:
[image]
The cable between Arduino Board and IMU Board is about 10cm long.
I tried with and without pullups resistors.
But same results.
Any ideea?
THX

richardbarnett, 2015-02-01:
You have I2C analyzers - nice! The traces show a read of the FIFO count which is zero as the IMU hasn't been set up. Is there any chance you could trap the very first access, which is a write to 0x6b? That's the access that is failing. Really, the sketch should not continue on to polling the IMU if it hasn't been set up but it does right now :-(.

iflavian, 2015-02-02:
Solved!
The problem was produced due to a cold welding to the GND n the IMU board. 
So the GND was floating...
I used the oscilloscope to look at the I2C signals and then with a magnifying glass and I discovered the problem. 
I reworked the welding and now the IMU is working very good with the RTIMULib.
Thank you very much for your great work and support.

closed 2015-02-02 by iflavian