richards-tech/RTIMULib-Arduino #14: yaw behaviour

nandomtl, 2015-02-05:
This is more of a question than an issue:
I have seen various demos of IMUs, and they seem a lot more stable than mine.
I have an Invensense 9250 (Drotek board: http://tinyurl.com/o85jeks ).
First, I ran the compass calibration sketch ArduinoMagCal.ino
Then, I used ArduinoIMU.ino to get the data from the IMU. I have a Processing sketch running on a Mac, receiving the data and displaying a block with an arrow (modified from the Razor package - same as this: http://tinyurl.com/nsmk7sm ).
test #1: board on a flat desk. I turn it 180 degrees, reading says it turned about 60 degrees.
test #2: pitching up 90 deg gives me a reading of 85 deg
test #3: pitching up 60 deg causes a yaw to the left from 5 to 10 degrees. Pitching down does the opposite.
test #4: roll left 40 deg causes a yaw of 10 to 20 deg.
Otherwise, pitch and roll behave in a fairly normal fashion. Yaw looks "drunk"...
Drift seems ok - even if there is jitter - but I am leaving it running overnight...
I downloaded the GitHub zip file on Jan 30, so the code is quite recent.

Is this normal for the hw+sw I am using? Or should I suspect I have a bad board?

Thank you in advance for your insight

richardbarnett, 2015-02-05:
Interesting. It doesn't seem as though the compass is actually calibrated - what you are seeing is fairly typical of an uncalibrated MPU-9250. When ArduinoIMU.ino starts up, does it report that it is using compass calibration? It will say one way or the other. Meanwhile, I will check my board to see if this is a software problem. I doubt very much that it is a problem with your board.

nandomtl, 2015-02-05:
Hello,

I left the board running overnight, zero drift, all axis.

Here's a sample of the Arduino consolde msgs:

ArduinoIMU starting using device MPU-9250
Using compass calibration
Pose: roll:2.65 pitch:-1.75 yaw:55.60
Pose: roll:2.58 pitch:-1.92 yaw:55.82
Pose: roll:2.54 pitch:-2.01 yaw:55.98
Pose: roll:2.53 pitch:-2.05 yaw:55.99
Pose: roll:2.53 pitch:-2.06 yaw:55.85
Pose: roll:2.55 pitch:-2.04 yaw:55.75
Pose: roll:2.57 pitch:-2.01 yaw:55.80
Pose: roll:2.60 pitch:-1.97 yaw:55.98
Pose: roll:2.62 pitch:-1.94 yaw:55.91
Sample rate: 85, calculating gyro bias
Pose: roll:2.63 pitch:-1.90 yaw:55.95
Pose: roll:2.66 pitch:-1.86 yaw:55.84
Pose: roll:2.67 pitch:-1.83 yaw:55.79
Pose: roll:2.70 pitch:-1.79 yaw:55.74
Pose: roll:2.71 pitch:-1.75 yaw:55.72
Pose: roll:2.72 pitch:-1.72 yaw:55.71
Pose: roll:2.74 pitch:-1.70 yaw:55.80
Pose: roll:2.75 pitch:-1.67 yaw:55.75
Pose: roll:2.76 pitch:-1.65 yaw:55.63
Sample rate: 84, calculating gyro bias
Pose: roll:2.77 pitch:-1.63 yaw:55.45
Pose: roll:2.78 pitch:-1.62 yaw:55.39
Pose: roll:2.79 pitch:-1.60 yaw:55.15
Pose: roll:2.78 pitch:-1.60 yaw:55.11
Pose: roll:2.78 pitch:-1.59 yaw:55.05
Pose: roll:2.78 pitch:-1.58 yaw:55.38
Pose: roll:2.79 pitch:-1.56 yaw:55.73
Pose: roll:2.79 pitch:-1.56 yaw:55.93
Pose: roll:2.79 pitch:-1.55 yaw:55.77
Sample rate: 84, calculating gyro bias
Pose: roll:2.79 pitch:-1.55 yaw:55.79
Pose: roll:2.79 pitch:-1.56 yaw:55.84
Pose: roll:2.80 pitch:-1.55 yaw:55.64
Pose: roll:2.80 pitch:-1.54 yaw:55.84
Pose: roll:2.81 pitch:-1.53 yaw:55.59
Pose: roll:2.81 pitch:-1.53 yaw:55.50
Pose: roll:2.81 pitch:-1.53 yaw:55.59
Pose: roll:2.81 pitch:-1.54 yaw:55.71
Pose: roll:2.81 pitch:-1.54 yaw:55.79
Pose: roll:2.81 pitch:-1.54 yaw:55.96
Sample rate: 84, calculating gyro bias
Pose: roll:2.81 pitch:-1.54 yaw:55.85
Pose: roll:2.80 pitch:-1.54 yaw:56.01
Pose: roll:2.80 pitch:-1.54 yaw:56.10
Pose: roll:2.80 pitch:-1.54 yaw:55.99

richardbarnett, 2015-02-05:
I have now checked this on my system and it works fine. I suggest that you redo the compass calibration. Two things are very important when you do it:

1. You must ensure that each axis has achieved its maximum value - this can be done by waggling the board around in each of the 6 main orientations (x up, x down, y up, y down, z up, z down). Make sure that you have saved the configuration (it's easy to forget). The display will show a confirmation when this has been done.

2. Make sure you are not near a magnet! Many desks are highly magnetic, as are many keyboards and monitors. Try to do the calibration in free space and do the same when you are testing. If the IMU is to be mounted in something that could affect this, make sure that calibration is performed after mounting in the device (i.e. you have to spin the whole device around to hit the six axis minima and maxima, not just the IMU).

richardbarnett, 2015-02-05:
Regarding the noise, this file - https://github.com/richards-tech/RTIMULib-Arduino/blob/master/libraries/RTIMULib/RTFusionRTQF.cpp has two constants, the R and Q values. These control the extent of the filtering. The basic setting is for quick response but it does allow sensor noise through. By changing these values you can vary the effect.

richardbarnett, 2015-02-05:
You can also change the low pass filter settings by changing the defaults in this file - https://github.com/richards-tech/RTIMULib-Arduino/blob/master/libraries/RTIMULib/RTIMUSettings.cpp

nandomtl, 2015-02-06:
Thanks for the suggestions.

Re-calibrating improved the yaw response when only yaw changes.
I still have lots of yaw associated to both pitch and roll.
Note: wooden desk and a Mac (aluminium)...

I tried tweaking the parameters you suggested: 
Modifying Q and R caused interesting change in behaviour.
The original values in my file were:
Q=0.001f
R=0.0005f
The best results I got were with R=50*Q, instead of Q=2*R, such as:
Q=0.001f
R=0.05f
The trade off is that after many repetitive movements, returning the board to the original position provides readings different from what they should be. For instance, if I pitch up from zero to ~70 degrees, back to zero and repeat several times, pitch will change to gradually smaller values (-1, -2, ...). Clearly a cumulative error problem.
What values are you using for Q and R? The defaults Q=0.001f and R=0.0005f?

I'll try changing the LPF in RTIMUSettings.cpp tomorrow, if I don't have any bright ideas.

Thanks very much for your help, much appreciated.

richardbarnett, 2015-02-06:
What's happening is that you are basically telling it to completely ignore the accelerometers and magnetometers. Any time you get a cumulative error means that just the gyros are being used and the error is caused by drift and integration errors. You would see the same effect if you just told the fusion algorithm not to use the accels and mags!

Something you could try if you wanted is to use RTHostIMU (or RTHostIMUGL is you want the OpenGL visualization) from the RTIMULib repo. You load the RTArduLinkIMU sketch on the Arduino and run RTHostIMU on a desktop. The advantage of this is that you get better calibration resources including ellipsoid. If after getting good min/max data you still see significantly changing yaw when changing pitch and roll, it might be worth trying the ellipsoid fitting calibration just to see what happens.

nandomtl, 2015-02-06:
Thanks for the explanation - I did the changes blindly, without fully understanding what happens under the covers. 

I did think of using the Host code - but my objective is to have the arduino+IMU independent from any other device (such as a desktop computer).
As you suggest, I might try it just to see what happens, and perhaps have an idea on how to fix.

Right now I am thinking of perhaps trying another board.

And I thought this would be a straightforward thing...

closed 2015-04-01 by richardbarnett