richards-tech/RTIMULib-Arduino #15: MPU9250 - Port onto TI Board

a2kram, 2015-03-07:
Hi!

Thank you so much for your effort, I tried running your library with an MPU9250 on an Arduino Mega and got excellent results! There is virtually no drift, something I couldnt really solve with the Kalman filter algorithm I wrote myself. Anyways, I was planning to use this filter algorithm on a TI ARM Cortex M4 board, and I had a couple of questions if you wouldnt mind:

1) What you say there would be any changes needed in the IMURead function? It seems there is a tight control over sampling rate, something which might need tuning on a non-Arduino board I would presume? 

2) I am trying to understand your calibration process, so you store the max and min values for the magnetometer along the three axes? Why is it that the offset and scaling factor are computed not during calibration but later on? Also, what is the purpose of CALLIB_DATA_VALID_LOW and HIGH? Is it just to make sure that reads from double e are correct?

3) Out of curiosity, if you are using quaternions in your filter, why is pitch limited to a 180 degree range? 

Again, thank you very much for your effort and great work!

Akram

richardbarnett, 2015-03-07:
Here are my answers:

1. You would need to replace I2Cdev with something appropriate to your processor. If you kept the calls looking the same, no other changes would be necessary. The sampling rate is set by the MPU-9250 in fact so this should not be an issue.

2. Yes, min and max on all three axes. The offset and scaling calculation could be done at either point. And you are right - those valid symbols are used to check for valid data in the EEPROM.

3. It's just the way it works. It has nothing to do with using quaternions or not using quaternions. The ranges are such that all possible poses in three dimensional space are covered.

a2kram, 2015-03-08:
Thanks Richard for your reply. I had a couple more questions.

1) The yaw, pitch, and roll provided, are they using aero-convention (yaw then pitch then roll)?

2) Can you elaborate on what you mean by all possible poses? How would you deal with a case where the pitch goes up to 90 degrees then instead of continuing to 180 degrees, goes down again to 0. Wouldnt this circumstance be ambiguous? 

3) Would you expect any funny behavior if the filter is not updated instantly? I tried adding a delay of 10 up to 30 ms to simulate processing time after the fusion.newIMUData call to see if any issues may arise, and I did not notice much issues. 

Thank you for your response.

Akram

richardbarnett, 2015-03-08:
1) It's standard quaternion -> Euler angle processing.

2) No. The yaw flips by 180 degrees when the device goes past vertical. Incidentally, this is why it's best to work in the quaternion world for as long as possible - to avoid these discontinuities.

3) Not sure what you mean. The data from the sensor driver is timestamped so it doesn't matter if there are delays after this - the fusion algorithm always uses the timestamp in the data, not the time when the data arrived at the fusion algorithm.

a2kram, 2015-03-10:
A few more clarifications please.

1) I currently have the library performing very well. However, sometimes I notice that the yaw jumps to an incorrect value then slowly returns to the correct range. This only happens to the yaw measurements, pitch and roll look very good so I am assuming it must be something to do with the compass readings. Have you noticed anything like that before?

2) The yaw, pitch, and roll measurements dont seem to go fully till +-180 or +-90. For example, the pitch seems to go from -83 to +83 only, is there a way to correct this through calibration? I tried to follow your instructions when calibrating the magnetometer.

Thanks!

Akram

richardbarnett, 2015-03-10:
1) No - that's very odd. Maybe you could record the raw magnetometer readings when that happens to correlate something with the jump?

2) It's possible that you also need accelerometer calibration. This isn't included in the repo as it usually isn't a major problem. Again, it might be worth looking at the raw accelerometer readings to see how they are behaving and when they get to or exceed 1G.

a2kram, 2015-03-11:
Thank you for your response Richard. Would you know of any good method to calibrate the accelerometer? I am hoping having to rotate the sensor in all 6 directions and computing offsets/scaling factors wont be necessary lol.

Also, I successfully ported your code to work on the TI CC3200 board. I had to rewrite the code fully in C so no classes or function overloading, but I would love to send it to you if you would like. I refactored the code a bit so I no longer have some of the original variable names (I try to keep them generic, programmers OCD haha).

richardbarnett, 2015-03-11:
Accelerometer calibration is a total pain. My technique is described in this document - https://github.com/richards-tech/RTIMULib/blob/master/Calibration.pdf <https://github.com/richards-tech/RTIMULib/blob/master/Calibration.pdf>. There is also code for Linux - https://github.com/richards-tech/RTIMULib/blob/master/Linux/RTIMULibCal/RTIMULibCal.cpp <https://github.com/richards-tech/RTIMULib/blob/master/Linux/RTIMULibCal/RTIMULibCal.cpp>.

However, it is difficult to get right and very easy to get wrong which will make things worse.

Glad you got the port to work. I can't really do anything with your code but you might want to post on GitHub or something (assuming that you haven't already done this) in case others might like to use it.

Richard.

---------------------------------------
Richard Barnett
www.richards-tech.com

a2kram, 2015-03-12:
Ok, thanks for your help Richard!

a2kram, 2015-03-17:
Hi Richard, 

I am having new issues when I try using multiple IMUs. If I use a single IMU, values are perfect with no drift and very consistent readings even with rapid motion. If I use two IMUs, same thing, no problems at all. If I use three IMUs however, one of the IMUs starts having very weird yaw behavior. I tried making my code as efficient as possible in order to make sure that between IMU reads and updates there is as little latency as possible but there wasnt much improvement.

I am curious to know, if an IMU reads and processes data, how long till the next read and process do you expect it to work correctly? Like if there is a 30 ms interval for example, would that affect the IMU results? Also, if the interval in between reads and processes is irregular, do you think that would cause any problem? Like if consecutive reads to one IMU are 30-45 ms apart, but not consistently the same interval, do you think that may cause an issue?

Thanks,

Akram

richardbarnett, 2015-03-18:
Regarding the sampling interval, for gyros and accels there is not a problem as the time interval is calculated from the sample rate - the IMU chip puts readings into the FIFO at defined intervals. The magnetometer is different as its data does not go into the FIFO. The software reads registers that so if the rate at which the processor reads samples is way behind the sample rate, the magnetometer readings for all samples read in one access will all be the same. I don't know if this explains what you are seeing, however, as it doesn't really explain why only one is affected.

What sample rate are you using? Also, are you seeing any FIFO overflows? I am wondering if this problem is actually caused by overloading the CPU.

a2kram, 2015-03-19:
Hi Richard,

I checked if the FIFO's are being overflown, or have size == 1024, and that never seems to happen. Also, the sampling frequency was originally 80 for both compass and MPU, then I dropped the compass back to 40, then I tried dropping both to 40, but I dont notice a difference in behavior.

Like yaw would change with movement but if say I turn the sensor 90 degrees, it would go crazy before settling on like 5 degrees difference than starting value. I am starting to notice weird behavior with roll as well, so maybe it isnt a compass thing. 

Like you said, its interesting why only one sensor would be affected when three are connected at once. Have you ever had any issues connecting multiple sensors? I am running this code on a 80Mhz ARM Cortex M4 chip, that should be enough processing power no?

Thanks Richard!

Akram

richardbarnett, 2015-03-19:
How are you handling magnetometer calibration? Each IMU needs its own calibration data - maybe the third one is getting incorrect data.

a2kram, 2015-03-19:
Hi Richard, yeah I made sure of that. I have an array of IMU calibration data which I cycle through.

Another thing I tried is to flip the order by which I get the imu data, and now what was previously the first imu became third and worked fine. Only the first IMU to run has bad data.

Akram

richardbarnett, 2015-03-19:
So something specific to that IMU?

a2kram, 2015-03-19:
No, I meant to say that whichever IMU runs first has bad data. When I run them in a different order, its still always the first IMU to run that has bad data, no matter which actual IMU it is.

richardbarnett, 2015-03-19:
Something funny in the software then perhaps.

closed 2015-03-31 by a2kram