richards-tech/RTIMULib-Arduino #18: Yaw drift when moving right/left/forward/back

isaacdrafton, 2015-03-28:
Hello and good job with the library!

I'm trying to use the heading/yaw values from the MinIMU-9 v3 to drive a robot car straight and turn at certain angles.

Here's the steps I follow:
- connect the IMU to a breadboard and the breadboard to an Arduino
- calibrate the magnetometer with ArduinoMagCal and save the values
- upload the ArduinoIMU sketch
- move the breadboard around in a straight line, keeping the same heading
- observe the output

The roll and pith remain close to 0 degrees - all good.
The yaw is what I'm interested in and it's drifting a lot when I move the board around and physically keeping the same heading. There are no magnetic interferences around and I also tried testing on the floor, moving the breadboard in a straight line.
While testing, if I return to the initial place, I can see the value from the first reading, so it's drifting back.

Do you have any idea on how to solve this?
Thanks!

richardbarnett, 2015-03-28:
Well you certainly did some good testing - it really helps to understand what's happening so thanks for that. By moving the device in a straight line pretty much eliminates any questions about poor calibration. The fact that you are seeing yaw changes even then is interesting. I have to say, the only thing I can imagine is that there is a serious magnetic disturbance in the building - not just something small nearby. Something worth trying is to use a compass (which could be a compass app on a smartphone) and do the same experiment - move the phone in a straight line and see if the yaw changes. If it does, you have something weird going on magnetically. It if doesn't, I have no ideas!

richardbarnett, 2015-03-28:
By the way, if you don't need an absolute heading reference and can tolerate a long term drift, you could just turn off the magnetometers. That would be good for going straight and turning at known angles - you'd just lose the reference to magnetic north.

isaacdrafton, 2015-03-28:
Incredible. This was the last thing I would've thought. I tested with the
phone and it's indeed a problem with the building.
How can I disable the magnetometer? I only need absolute values.
Thanks!

richardbarnett, 2015-03-28:
In line 74 of ArduinoIMU.ino set true to false:

    fusion.setCompassEnable(false);

isaacdrafton, 2015-03-28:
The drift is highly reduced now but it's still not usable in my project
because of not so accurate output. Is there a way to improve or calibrate
the acc and gyro?

richardbarnett, 2015-03-28:
Since you are referring to yaw then the accels are not a factor. I am surprised that you are seeing a lot of drift - once the gyros have a valid bias correction they tend to be fairly stable. What drift rate are you seeing? BTW you may see a high drift rate unless the IMU has been stationary for about 5 seconds. If it is reset then in constant motion, it's impossible to for it to compute the gyro bias.

isaacdrafton, 2015-03-28:
I was testing with the IMU on the breadboard as before and waited for a
couple of seconds before doing anything.

Holding the breadboard stationary and observing the output:
- roll and pitch are slightly varying but in a constant interval.
- yaw starts at e.g. "-1.5" and drifts over time to 0 but as time passes,
the drift speed slows down.

When rotating the breadboard at known angles - Test 1:
- rotating 90 degrees counterclockwise: yaw goes from 0 to -84 then drifts
in around 1 minute to -45 and keeps increasing.
- rotating the board to the initial position: yaw goes from -45 to around
30 and keeps drifting to 0.

Test 2:
- rotating 90 degrees clockwise: yaw goes from 0.5 to 85 and then drifts
towards 0 as in the previous test.

Test 3:
- rotating 180 degrees clockwise: yaw goes from around 0 to 167 then drifts
towards 0. (one degree every couple of seconds).

So it seems that everything I do, when I rotate in any direction, the yaw
drifts towards 0.

richardbarnett, 2015-03-28:
That sounds like a software bug. I'll do some testing here and get back to you shortly.

richardbarnett, 2015-03-28:
I am seeing the same effect. There's something a little weird going on in the fusion algorithm under these circumstances. I am still working on it but it could be a while as I am not too sure what's happening. In the meantime, you could try reducing the value of RTQF_QVALUE in RTFusionRTQF.cpp. This reduces the influence of the accelerometers and gives more to the gyros. The drift will be somewhat proportional to this value.

isaacdrafton, 2015-03-29:
I tried with different and smaller values for RTQF_QVALUE but the drift is still there and it accumulates if I make multiple +/- 90 degrees rotations. Also the inaccurate readings are still there (+/- ~86 instead of 90)

richardbarnett, 2015-03-29:
Ok, there are a number of things going on here. To get better tracking with just the gyros, a higher sample rate might well be needed. The Arduino does have limited processing power so there is a limit to this (about 150 samples per second with RTIMULib-Arduino). Whatever you do though, you will get some drift as there is no absolute reference.

On the issue of the steadily reducing yaw, that appears to be due to a cheat I did in the fusion algorithm to avoid properly SLERPing the quaternions. I am going to test out the correct (but more complex) way of doing this to see if it helps.

I am hoping to get a new version out sometime today with better gyro-only tracking and a fix for the reducing yaw problem.

richardbarnett, 2015-03-29:
Well the new version of RTIMULib-Arduino is up. It may help with the gyro tracking, especially if you change the sample rate to something higher than the default, but it has not fixed this problem where the yaw trends down to 0 when the compass is disabled in the fusion algorithm. Again, changing the new RTQF_SLERP_POWER to a lower value will help but it won't stop it. I am at a loss to understand exactly why this is happening but at least the whole calculation is more accurate now. I will continue to think about this but I can't guarantee a fix I am afraid.

isaacdrafton, 2015-03-30:
Thank you for your effort but unfortunately this didn't help me. The yaw is still drifting to 0 and the readings are not precise.

richardbarnett, 2015-03-30:
Yep, it's weird. I am still working on this - it's actually quite fascinating. Sorry it doesn't work for you at the moment.

richardbarnett, 2015-03-30:
Ok, it's fixed! It was due to an inaccurate but fast square root calculation. I replaced it with a standard square root and it works much better - no return to zero on yaw.

The accuracy will be improved too but I still don't know if it ill be what you want. It's probably as good as it's going to get now.

isaacdrafton, 2015-03-31:
Cool! There is no return to zero but here's another issue:

- Turning the IMU to 90 degrees slowly will show less than 90 degrees on yaw. The slower I turn to 90 degrees, the slower the yaw increases. I managed to get 28 degrees when I was expecting 90. This inaccuracy accumulates. For example if I turn clockwise to 90 degrees slowly, and then 90 degrees counterclockwise faster, I don't get back to 0 but a different value.

It seems that the yaw depends on speed of rotation of the IMU.
Any idea why this is happening or how to fix it?

richardbarnett, 2015-03-31:
You are absolutely right. Sorry you seem to be hitting all kinds of problems when the magnetometers are turned off! I was mostly concerned about high speed turns and didn't check really slow ones. I have a theory about this - I am looking into it now and will get back to you...

richardbarnett, 2015-03-31:
The gyro bias calculation was being fooled by the slow movement. So, I changed it to stop adjusting the bias after the initial assessment is complete. What this means is that you need to avoid slow rotations while the bias is being calculated. When valid bias is achieved (after 5 seconds if the IMU is perfectly still), the slow turns should work a lot better.

isaacdrafton, 2015-03-31:
Now it's really cool. I get usable values from the gyro and it's not speed dependent.

- I still need to figure out why 90 degrees turns get me +/- 85 and why I never reach +/-180 when I do two 90 degree turns in either direction (it only gets to +/- 174). Any idea why?
- What do you think I should configure to get maximum reliability, quick reaction time and precision without using the compass?
Thank you very much for your support!

richardbarnett, 2015-03-31:
A higher sample rate is always going to help. I'll take a look at your angle range issue - maybe the gyro rate scaling isn't quite correct for that IMU. Progress is being made at least!

richardbarnett, 2015-03-31:
I have to say that I am not seeing this. I am using an AltIMU v3 which is basically the same thing with a pressure sensor on it. I get exactly 90 degrees and also pretty close to 360 when it is fully rotated.

One thought - is your IMU mounted such that pitch and roll are close to 0? That's how I am testing this.

isaacdrafton, 2015-03-31:
I have to roll close to 0 and the pitch close to 180 because the IMU is placed upside down on the board.

richardbarnett, 2015-03-31:
You may find it more convenient to redefine the reference orientation in that case. Actually, I expect that the roll is 180 degrees as pitch can only go to 90. If you look in RTIMULibDefs.h you'll see a whole bunch of possible orientations. It may be that changing to RTIMU_XNORTH_YWEST works better for you.

It really shouldn't make much difference to the quaternions but Euler angles are a bit more tricky so it helps to have the correct setting.

isaacdrafton, 2015-04-01:
Maybe something is wrong with my board. I tried with the correct orientation.
I'll probably scale my values to 360 degrees.
Thank you for your support!

closed 2015-04-01 by isaacdrafton