richards-tech/RTIMULib-Arduino #20: Got gibberish in serial monitor using RTArduLinkIMU and would stop in few seconds.

YangWH, 2015-04-14:
Hi,

I am trying RTArduLinkIMU, but I got gibberish as below, and updating would stop in few seconds.
http://ppt.cc/ZPi7
Did I do anything wrong? 
I am using Arduino pro mini 5V,16M,ATmega328 + MPU6050 + HMC5883

Best.

richardbarnett, 2015-04-14:
What are you trying to do? RTArduLink drives binary data to software such as RTHostIMU. If you want something that you can look at, try ArduinoIMU.

YangWH, 2015-04-15:
Thanks for you instruction, I tried ArduinoIMU.
But updating still stop in few seconds.
I can't figure out what's going on there.
Do you have any idea?
Best.

richardbarnett, 2015-04-15:
Actually no. You could try looking at the I2C signals with a scope to see if there's anything funny there. What kind of wiring to the IMU are you using?

maziarzamani, 2015-04-15:
I have checked the I2C data. They are clean and i have tried to underclock the speed to double-check that as well.

I am using SPI from µC to IMU (500KHz).

Cheers
Maziar

YangWH, 2015-04-17:
I don't have scope to check now. But I think it might cause by both 3.3V and 5V on I2C bus. I simply add a resistor on the bus and it works fine. I will try level shifter later. Thanks a lot for your great help!

richardbarnett, 2015-04-17:
Excellent.

closed 2015-04-17 by richardbarnett