richards-tech/RTIMULib-Arduino #21: Yaw data is not right

phil384, 2015-04-15:
I am getting very accurate date for pitch and roll, but yaw does not seem to be correct at all. It varies between 30 and 50, and when the imu is moving very slow it reaches lower angles. Any advice? Thanks

richardbarnett, 2015-04-15:
As I always need to ask - have you calibrated the magnetometers? You didn't say what IMU you were using - that would be helpful to know.

phil384, 2015-04-15:
I am using the Adafruit 10-DOF IMU (L3GD20H, LSM303, BMP180). The example program they give does not incorporate all of the sensors so I needed a fusion program to get the best pose data. 

As far as I know the magnetometer has been calibrated properly. 

Separate question: While I am currently using an arduino uno, would this program work with an arduino yun?

Thanks!!

richardbarnett, 2015-04-15:
Even without calibration that IMU would work pretty well. Are you sure that there aren't any local magnetic disturbances? It should work.

As for the yun, I have not tested that but it probably would.

richardbarnett, 2015-04-15:
I just checked my Adafruit 10-dof and it does seem fine. Maybe redo the mag calibration. Make sure to get good maxima and minima in all the axes and then save the results. What you are seeing definitely looks like bad calibration data.

phil384, 2015-04-15:
I reran the calibration and it seemed to work pretty well, even worked for the Yun. Thanks for the help. In terms of calibration data is there a max and min I should be looking for to indicate a good run? Also is there a way to set a certain  orientation as 0 pitch 0 roll 0 yaw.

richardbarnett, 2015-04-16:
For that IMU, something like -50 to 50 would be reasonable. However, things like cables etc can be magnetic so it can distort this very significantly.

Regarding the orientation, take a look at these two posts:

https://richardstechnotes.wordpress.com/2014/11/18/rtimulib-now-supports-imu-axis-rotation/

https://richardstechnotes.wordpress.com/2014/11/20/using-quaternions-to-correct-imu-installation-offsets/

Hopefully that will help.

richardbarnett, 2015-04-16:
I should mention that for the Arduino, the reference orientation is set by uncommenting the appropriate setting in RTIMULibDefs.h. The blog post refers to the Linux version.

Bastich-666, 2015-04-22:
Hello

I am getting a similar problem and I am 100% certain it is to do with my compass calibration. The reason being that when I set fusion.setSlerpPower to 0 I get the correct angles e.g. 90 degree turn is reported as 90 degrees however as expected this suffers from drift. When I set the value to anything above 0 then a 90 degree turn is reported as nearer 130 degrees. No matter how many times I try to calibrate it I can not get it to report the expected results. 

Could this be down to the location where I am trying to calibrate the sensor? The compass on my phone works fine in this location but I am assuming this is because it is already pre calculated. 

These were my last calibration results:
  minX: -29.67 maxX: 63.21
  minY: -43.10 maxY: 47.08
  minZ: -32.56 maxZ: 53.10

Thanks

James

richardbarnett, 2015-04-22:
Which IMU?

Bastich-666, 2015-04-22:
Sorry. I am using the SparkFun MPU-9150.

richardbarnett, 2015-04-22:
Are you sure you are getting the real max and min? Those results look a bit symmetrical around zero. Usually the 9150 is all over the place. It's very important to move the IMU so that all maxima and minima are hit - just waving the IMU around will not work. I know some systems use a figure of 8 system but this doesn't. So it takes a bit of concentration to position the IMU with each axis up and them down (total 6) and waggle them about in each of the 6 positions because the extrema may not be totally aligned with the physical axes.

Silly question but you are remembering to press "s" and enter to save the calibration data? It does not save it automatically. ArduinoIMU reports at the start whether it is using calibration or not...

Bastich-666, 2015-04-22:
When I ran the calibration sketch I rotated the sensor around in all directions until the displayed data in the serial monitor had stopped updating. I then sent the 's' command and it reported that the data has been saved. The ArduinoIMU reports that it has calibration data although on one run it did report not having any at which point I recalibrated  it and it has not done this since. The previously posted Max and Min values I copied from the serial monitor once it had speed updating.

I will try calibrating it again and post my results. I will also try it in a different room via my laptop and an Arduino Nano instead of my desktop and an Arduino Uno.

Thanks for getting back to me so quickly.