richards-tech/RTIMULib-Arduino #22: send data to an SD card

phil384, 2015-04-20:
I wanted to save the fused imu data to an SD card?
what would be the proper command to get this data to a String:
"Pose: roll:-4.23 pitch:-0.75 yaw:74.22, pressure: 1477.74, temperature: 26.50"

I couldn't figure out which part of this command from the ArduinoIMU10 program would be useful:
RTMath::displayRollPitchYaw("Pose:", (RTVector3&)fusion.getFusionPose());

Thanks

richardbarnett, 2015-04-21:
You could do something like this:

RTVector3 pose = fusion.getFusionPose();
fd.println(pose.x());    // roll
fd.println(pose.y());    // pitch
fd.println(pose.z());    // yaw

Of course the print can be anything you want.

phil384, 2015-04-23:
that section of code seems to print values which are not scaled right. Am i missing a step?
Original code:Pose: roll:-3.69 pitch:-0.85 yaw:63.30
Modified code: roll:-0.06 pitch:	-0.01 yaw:1.10

richardbarnett, 2015-04-23:
Ah yes. The display routine converts radians to degrees whereas the pose.x(), y() and z() values are in radians. You can multiply by RTMATH_RAD_TO_DEGREE to convert radians to degrees. There's also RTMATH_DEGREE_TO_RAD that goes the other way.