richards-tech/RTIMULib-Arduino #24: Disabling EEPROM?

th0f, 2015-04-28:
Not really an issue, but hope you can help. 
I'm trying to get this working on an RFDuino, but this does not have any EEPROM. I looked at the code and it looks like the Arduino Due has an exception to use flash. Any ideas how to get this to work with an RFDuino or is there a way the EEPROM calibration data can be hard coded? 
Thanks!

richardbarnett, 2015-04-28:
Simplest way would be to hack this file:

https://github.com/richards-tech/RTIMULib-Arduino/blob/master/libraries/CalLib/CalLib.cpp

The calLibRead() function could be modified to populate the calData structure with hard coded values.

As for using flash as EEPROM, that is not really useful on the Due anyway as it gets erased on every upload. The RFDuino does has the ability to use flash:

https://github.com/RFduino/RFduino/blob/master/libraries/RFduinoNonBLE/examples/Flash/FlashStructure/FlashStructure.ino

So another way would be to modify CalLib.cpp to use these functions instead of the EEPROM calls. Don't know if the data will survive an upload though.

th0f, 2015-05-13:
Got it working by changing CalLib.cpp:

```cpp
#ifdef RFDUINO
// RFduino version

void calLibErase(byte device)
{
   
}

void calLibWrite(byte device, CALLIB_DATA *calData)
{
    
}

boolean calLibRead(byte device, CALLIB_DATA *calData)
{
    calData->validL = CALLIB_DATA_VALID_LOW;
    calData->validH = CALLIB_DATA_VALID_HIGH;
    calData->magValid = true;
//    calData.pad = calData.pad;
    calData->magMin[0] = -35.05; //x
    calData->magMin[1] = -52.18; //y
    calData->magMin[2] = -28.62; //z
    calData->magMax[0] = 64.28; //x
    calData->magMax[1] = 60.31; //y
    calData->magMax[2] = 66.83; //z
    return true;
}

#elif __SAM3X8E__
// Due version

```

For anyone looking for a same solution, simply replace 
```cpp
#if __SAM3X8E__
// Due version
```

in the original file for the above code.

closed 2015-05-13 by th0f