richards-tech/RTIMULib-Arduino #25: IMU set up uses too much memory in Arduino

Prandr, 2015-04-30:
```C++
#include <Wire.h>
#include "I2Cdev.h"
#include "RTIMUSettings.h"
#include "RTIMU.h"
#include "RTIMUMPU9150.h" 
#include "CalLib.h"
#include "EEPROM.h"
RTIMU *imu;                                           // the IMU object
RTIMUSettings settings;                               // the settings object
void setup() {
  // put your setup code here, to run once:
 imu = RTIMU::createIMU(&settings); 
 }
void loop() {}
```
The problem is that just the very basic code above occupies 47% of program memory with Arduino 1.6.3 compared to 18% if i comment the line, That means it eats whole  29% (!), which is quite puzzling, because I looked in the code: all it does is copying a pointer and a few bools. I checked it many times.
Thank you for your assistance

richardbarnett, 2015-04-30:
I can only assume that if you comment out that line of code, the linker doesn't bother linking most of the library.

Prandr, 2015-05-06:
why should it link whole library, if I only create an object without using any methods?
Anyway I would like (or rather required) to use a TFT display, which itself typically consumes 40-50% of storage. Is it possible to reduce amount of resources. used by RTIMU? 
All i need is get raw (unfused) data from sensors, transfer to PC for processing and then transfer back to display on the TFT. I have MPU-6050+HMC5883L

richardbarnett, 2015-05-06:
If you wanted the smallest footprint and no fusion, you could just look at RTIMUMPU9150.cpp (it supports the 6050 + 5883 combo now) and take the bits you need from that. Or else I am sure that you could find some basic drivers on the web somewhere - Sparkfun or Adafruit probably have something. RTIMULib is designed to do a lot more than you need.

Prandr, 2015-05-06:
Thank you for your reply!
Actually on the other hand it is part of the problem, that I can't use advanced features of RTIMU, because they leave no room for something else, like TFT, and I therefore have to implement them on the PC.

richardbarnett, 2015-05-06:
If you want to stick with Arduino, what about the Mega 2650? That would solve the flash problem.

Prandr, 2015-05-06:
I am actually solving a student project, and have (or even required) to use hardware provided by the univercity-  that is Pro Micro/Leonardo...
Ok as I see, there is no way but perform more or less complex processing on the PC. But still it would be nice to reduce the memory usage. May be you could give a hint, which library functions could be truncated (omitted from it) under condtion that I use IMUinit(), IMURead(), and getGyro/getCompass/getAccel. that is that are not in use by this functions.

richardbarnett, 2015-05-06:
First off is to make sure that all the fusion related stuff is removed. I would delete RTFusion.h and .cpp, RTFusionRTQF.h and .cpp. You could also delete all the quaternion related stuff in RTMath.h and .cpp.

I would also delete all the unused files from RTIMULib. All the code for devices other than you are using should be deleted. All the pressure stuff. They shouldn't be being used but you never now. Then see if that helps on the memory front.

Prandr, 2015-05-07:
Unfortunately this has changed nothing.
However, I found some unnessary code in ccompass configuration, that checks if and what type of magnetometer is present. Since the hardware is known and we don't require interoperabilty, it can be trimmed, which saved 2% of memory. Could you please verify that I haven't thrown out anything important?
```C++

bool RTIMUMPU9150::configureCompass()
{
    unsigned char asa[3];
    unsigned char id;

    m_compassIs5883 = true;
    m_compassDataLength = 6;
	m_compassPresent=true;
    
    if (!bypassOn())
        return -11;

        // HMC5883 is present - use that

        if (!I2Cdev::writeByte(HMC5883_ADDRESS, HMC5883_CONFIG_A, 0x38)) {
            bypassOff();
            return -12;
        }

        if (!I2Cdev::writeByte(HMC5883_ADDRESS, HMC5883_CONFIG_B, 0x20)) {
            bypassOff();
            return -12;
        }

        if (!I2Cdev::writeByte(HMC5883_ADDRESS, HMC5883_MODE, 0x00)) {
            bypassOff();
            return -12;
        }

    if (!bypassOff())
        return -16;

    //  now set up MPU9150 to talk to the compass chip

    if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_I2C_MST_CTRL, 0x40))
        return -17;

        if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_I2C_SLV0_ADDR, 0x80 | HMC5883_ADDRESS))
            return -18;

        if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_I2C_SLV0_REG, HMC5883_DATA_X_HI))
            return -19;

        if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_I2C_SLV0_CTRL, 0x86))
            return -20;
    

    if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_I2C_MST_DELAY_CTRL, 0x3))
        return -25;

    if (!I2Cdev::writeByte(m_slaveAddr, MPU9150_YG_OFFS_TC, 0x80))
        return -26;

    if (!setCompassRate())
        return -27;

    return 1;
}

```
thanks for your help!

richardbarnett, 2015-05-07:
The question is - does it still work? :-) Yes, there was code to handle the 9150 compass that would not be needed.

Prandr, 2015-05-07:
Yes, it compiles, but if works correctly and delivers proper values  - that's the other question...

Prandr, 2015-05-13:
I recovered whole 14% of memory by getting rid of inheritance. That is transferring all attributes and methods from RTIMU to to RTIMUMPU9150.