richards-tech/RTIMULib-Arduino #27: how does the ”if(++loopCount >=10);continue; " loop in the examples work?

TurbulentCalm, 2015-05-11:
These loops, in the examples, have me baffled.

The code:

    int loopCount = 0;
    
    while (imu->IMURead()) {   // get the latest data if ready yet
      // this flushes remaining data in case we are falling behind
      if (++loopCount >= 10)
        continue;

How does loopCount get reset?

The way I see this working, is to skip the loop for ten increments, displaying IMU data and then just loop back to the

        while (imu->IMURead()) 

statement forever.

As this does not happen, how does loopCount get reset?

TurbulentCalm, 2015-05-11:
Arg! Nothing like posting a question to sharpen the mind!

Now I see it. The loop() function in Arduino code tripped me up.

    • Read the first ten IMU data (if there is that many to read)
    • now quickly read any remaining data until no IMU data is left to read
    • fall out of while loop but start code again due to the property of loop() function
    • set loopCounter to zero again
    • rinse and repeat...

Why was that so hard to see?!? ;)

richardbarnett, 2015-05-11:
Well it's probably not the most elegant bit of code I have ever written!

closed 2015-05-11 by richardbarnett

TurbulentCalm, 2015-05-11:
I don't know, it's small and it works... ;)

TurbulentCalm, 2015-05-11:
Richard 

Hi and thanks for your library. I got the old one first then found your updated one. Much smaller, simpler and nicer. Taken learning it step by step.

I'm building a robot with Python code on a Raspberry Pi 2 and two Arduinos. One plugs onto the Pi, which is handy but also a pain as I can't access the Pi's GPIO pins (I'm using Freetronics PiLeven for that) and the other connects via USB (it's a Freetronics Eleven). Both these both boards are Arduino clones.

I tried using the IMU directly through the PiLeven's I2C port and a version of RTIMULib (?) for Python, but I could never get the Pi to recognise the IMU, even after I cut the links to the 10k pull-ups (if connected directly to the Pi it worked fine, even with the IMU pull-ups). In the end I decided to run it on the USB connected Eleven board, which is now the Sensors Arduino and use the PiLeven to run motors and relays etc.

After a disastrous attempt to use PyMata. I started from scratch and wrote my own serial communications class and command handler classes. Lots of hard work and hours staring at code that 'should work' but doesn't but now it's finally paying off. The robot has two drive motors, 1 ultrasonic distance sensor (soon to be 5, when I get around to building the connection, ribbon cables, ½ hour job, but I just can't get it done) and the IMU.

I'll be adding a head (articulated with two servos) carrying a webcam, light sensor, a sound pressure sensor and a small speaker, connected to the Pi. Two IR reflectance sensors, which I hope will help detect objects about to hit a drive wheel having got past the US distance sensors (maybe). It will also have a small piezo electric speaker as well, drive by one of the Arduinos, for general error warning signals. I should just have enough IO pins on the two boards to do all that.

It also monitors it's battery voltage level, while on batteries, and shuts down the Pi, switching off the power after a set delay (sort of, my custom circuit needs a redesign, but it will not drop power if main power is turned off, until the Pi has shutdown, which is good enough for now).

I'm considering post the code on GitHub latter and maybe writing an Instructable regarding the robot's general construction and how to tailor the code to fit any need.

I've attached some photos. Hope I didn't bore you. Keep up the great work.

See photos of the robot via my iCloud link: https://www.icloud.com/photostream/#A25ON9t3bBBJa

Regards

Paul Greeve

richardbarnett, 2015-05-11:
Paul,

Looks like a fun project - it's going to be a complicated system in the end!

Richard.
-------------------------------
Richard Barnett
www.richards-tech.com