richards-tech/RTIMULib-Arduino #29: Pitch problem

maxouille, 2015-06-01:
Hi guys, 

I have some issue with this library. 
My pitch value retrieved from the fusion.getFusionPose() is weird. 

The value behave like this when I made a rotation of 360 degrees : 0 -> 90 -> 0 -> -90 -> 0. 

The other values (roll and yaw) behave like this : 0->180 -> -180 -> 0. 

It is quite annoying because when I draw a cube with the pose, his behaviour is weird. 

Thx for the help !

richardbarnett, 2015-06-01:
That's exactly how it is supposed to work:

http://en.wikipedia.org/wiki/Euler_angles#Signs_and_ranges

If you are seeing the cube with incorrect pose, this is generally because the rotations are being applied in the wrong order. It is not always obvious what the correct order should be and might require some trial and error.

maxouille, 2015-06-01:
Oh I have to apply rotations in a specific order ?

And another question, may be it is wrote in the wiki page but when I move around the pitch axis, yaw is moving too. Is it normal ?

Thx for this awesome lib btw !

richardbarnett, 2015-06-01:
Yes.

Did you calibrate the magnetometers? If you are using an MPU-9150 or MPU-9250 it is absolutely essential but needs to be done for all IMUs to get decent results.

maxouille, 2015-06-02:
I changed my rotation order to Z, X and Y and it almost works.
Moreover, I use your Calibration Script to calibrate the magnetometer. The orientation in the real world is working except when the pitch value is +- PI/2 rads. At this value, yaw move too and I don't understand why the algorithm do that. 

Thx a lot for your help.